JaredC01/Galileo2 #16: BBT 2209 USB C port is hitting the motor.

Tonykostansek, 2023-11-13:
Am i doing something wrong.  The 2209 board wont fit with the usb C port.

JaredC01, 2023-11-15:
I'd say you're probably doing something wrong, yes.  The 2209 and 2240 boards fit with JUST enough clearance.  If you have a picture of your setup, I can take a look, but Discord is a better place for troubleshooting since it can be done quickly.

Tonykostansek, 2023-11-15:
Okay ill take some photos and Update you on Discord.

Thanks

Tonykostansek, 2023-11-28:
I have an update on my issue.  I reprinted all the extruder parts using my new Bamboo X1. and all the part fit better and the UsbC port though still very close does fit now. 
Thanks for the reply Jared.

T

jinnn1989, 2023-12-07:
Link to the discord post ? I am having issues mounting the PG7 gland, not sure how to connect it. Some photos would really help

closed 2024-03-25 by JaredC01