JaredC01/Galileo2 #17: Extruder motor testing

kyper311, 2023-11-15:
Is there a way to test if the motor is working. New to Voron commuinity. Using sb2209 canbus all other features of extruder working correctly, ie fans, temps etc. I am unable to determine if the motor is spinning. Extruder will not extrude or retract. Any testing or RC you can suggest. If this is not the correct forum for this please let me know a good source to reach out to for assistance

austinrdennis, 2023-11-15:
If I were in your position, I'd remove the nozzle and shine a flashlight up the heat block while you look down through from the top. Then you could see if it was spinning pretty easily

kyper311, 2023-11-15:
not sure i  understand the suggestion. If you remove the nozzel and look into the filiment path from top down there is no way to see the little gear moving that i can see. i have taken the extruder apart and trying to see if there is a way with just the motor connected to see if is tunring. I can move the gear box freely by hand so there is no blocking and i can turn the motor gear with fingers easily there is no locking.  Is there anyway with the extruder apart to provide power to see if it turns?

JaredC01, 2023-11-15:
@kyper311 your best bet would be to join the Discord and jump into the Galileo channels there for troubleshooting.  Troubleshooting via GitHub is significantly more difficult.

kyper311, 2023-11-15:
Hi Jared..

Great name by the way.. really appreciate you getting back to me so
quickly. I rewatched your 5 hour video with Steve and realized I had to
configure the current and rotation and after checked omhs realized the
black wire was pretty loose.. I rebuilt the extruder and its running now..
I really appreciate your advise and will engage with the voron discord for
future trouble shooting.. I was not sure where to start so thought best to
go right to source designer..

Again.. really glad you took the time to reach out to me. I really
appreciate it

Take care
Jared

closed 2023-12-23 by JaredC01