JaredC01/Galileo2 #18: Does the manual need an update for the new kit batch?

Bart97, 2023-11-15:
Hey.

As far as I understand the latest kit batch includes injection molded plastic parts instead of MJF. Does that mean that running-in the gears with a drill is no longer needed (page 28 in the manual)?

JaredC01, 2023-11-15:
That's an accurate statement, yes.  I'll be working on a revised manual for the injection parts to avoid any confusion prior to full release.

Bart97, 2023-11-15:
> That's an accurate statement, yes. I'll be working on a revised manual for the injection parts to avoid any confusion prior to full release.

Thanks for the clarification :)

Bart97, 2023-11-15:
> That's an accurate statement, yes. I'll be working on a revised manual for the injection parts to avoid any confusion prior to full release.

And is the silicone grease still the preferred lube for the plastic parts?

djeZo888, 2023-11-16:
So, we should use v1 manual for assembling this one? I've been searching for any kind of manual online, even a YT video...
Also, a list of recommended greases would be welcome, need to cross check what I have and what I need to buy. Thanks!

djeZo888, 2023-11-16:
I'd like to know regarding which grease (or even better, a list of recommended greases) to use as well, it is not mentioned. Thanks!

deremka, 2023-12-02:
Yes, thats sufficient. However if you want longterm greasing its better to use Lithium/PAO Grease as silicone sticks less:  options include Molykote EM-30L (my preferred for Plastik-Gears) and SuperLube (in Europe called SyncoLube) Multi-Purpose Synthetic Grease (warning they have different Lubes with slightly different Names and Properties, you want exactly this). The Superlube is the most versatile and frequently used in 3D Printing, but do not use for Linear Rails!

rogery555, 2023-12-18:
One of the best greases for this application which is readily available would be Dow Corning Molykote EM-30L

closed 2024-06-26 by JaredC01