JaredC01/Galileo2 #21: New Batch Tolerances

deremka, 2023-12-02:
received my new Kit today which has 2 Issues with tolerances

1. the hole in the injection molded gear housing is very slightly to small for the Stepper and needs some slight sanding and cleaning before inserting the gears. Otherwise its very difficult to make it a drop-in insert
2. The bearings for the Planets don't fit on the 3 aluminium axes of the shaft, as they were to wide. It took me 2h of sanding before I could fit them.

Just FYI

StefanX13, 2023-12-09:
2nd was true for me as well.

JKtech3D, 2023-12-12:
Inserting the bearings into the planetary gears was a tight fit for me, but everything else was fine.

kaibsora, 2023-12-20:
i had the first issue and just ended up press fitting it into place before putting the planetary gears into place,

JaredC01, 2024-01-02:
The hole in the ring gear housing should be tight enough to require pressing onto the stepper; that is by design.  It should require a firm press to seat onto the stepper properly.

The carrier shaft has had a bit of variance in the past as well, and I think it's down to the tooling of the CNC machine getting a bit dull.  My suggestion for that would be to put the bearing onto the shaft without the injection planet, and use something like a 20T pulley with a set of parallel-jaws pliers to fully seat the bearing on the carrier.  I would NOT suggestion sanding if you can avoid it, as sanding can be imprecise and could potentially cause issues with planet alignment if you take too much off of one side.

closed 2024-03-25 by JaredC01