JaredC01/Galileo2 #26: SB2209 (RP2040) not enough clearance w/Cover

Ixian, 2023-12-28:
I attached photos as they show the problem clearly:

There are two versions of the SB2209 (not counting the version that uses a 2240 stepper) - the newer version based on the RP2040 MCU uses slightly larger connectors, which is helpful when it comes to crimping but doesn't leave any room for the cover. It won't fully close. Absent CAD is there any chance you could make a cover with slightly more room/curvature? I dug up a recent mod (this one: https://github.com/JoeCorelli/Galileo_2_SB2209_SB2240_MOD/blob/main/README.md) but that cut out part of the front body and I'd rather not depart from the core design if possible.

Thank you! 
[image]
[image]
[image]

karlfife, 2023-12-29:
Same issue here.  Ran into this exact problem months ago when trying to incorporate tow RP2040 board with the CW2  (The with the modified CW2 motor plate that receives the reinforced latch).     I too would love a solution that doesn't involve drills, hacksaws.  The EBB RP2040 is a great board.   I've also been noodling on a mod that would divert a very tiny amount of air from the hotend fan into the area where the EBB is housed.   Any amount of airflow greater than zero would server to cool the step driver, enable the on-board thermistor to be more representative of the chamber temperature, and knock a few degrees off the stepper itself--all without the mass and complexity of an additional fan.  Ironic that it's sitting next to a nice big fan. :-)

Shan723, 2023-12-29:
Same here but I have normal 2209 and it sticks out more preventing the front being closed all the way 
[image]

Ixian, 2023-12-29:
> Same here but I have normal 2209 and it sticks out more preventing the front being closed all the way [...]

No, you have a different problem. I have that version as well and just did a test fit, closes fine (I didn't screw it in so you can still see a tiny gap but trust me, it fits flush).

Not sure why yours is like that but the original SB2209, the one with 1.25mm pitch picoblade connectors, fits ok. Still tight and all of these boards with onboard extruders need some kind of venting or cooling but it works. The newer rp2040 model is the problem. 

I've hacked together a slightly larger (about 2mm inside, enough to keep cables from being squished) cover with hex pattern vents that works, though I can't get it to line up perfectly since I am hacking the stl. It will do until a proper solution, or the CAD for the G2E, is released.

Shan723, 2023-12-29:
I guess Ill make a separate thread

Ixian, 2023-12-29:
> I guess Ill make a separate thread 

I don't know that it is a problem a separate thread will fix - looks like something on your end. Perhaps the breakout board needs adjusting? Because mine fits fine.

Ixian, 2023-12-29:
> I guess Ill make a separate thread

Looking closer at your pic, I see what appears to be a socket head screw sticking out past the edge of the SB breakout board. You are also using socket heads to secure the MCU to the side of the extruder. Both those call for button-headed screws. On the MCU, likely doesn't matter but could be causing you a problem on the breakout. 

If it worked before for you w/CW2 then possibly that design was a little more forgiving; in any case I would replace those with the correct screws and insure the breakout is fully seated with nothing but the connector sticking past the edge.

karlfife, 2023-12-29:
> I've hacked together a slightly larger (about 2mm inside, enough to keep cables from being squished) cover [...]  It will do until a proper solution, or the CAD for the G2E, is released.

Would you be willing to share your (RP2040)-friendly door mod?

Ixian, 2023-12-29:
> > I've hacked together a slightly larger (about 2mm inside, enough to keep cables from being squished) cover [...]  It will do until a proper solution, or the CAD for the G2E, is released.
> 
> Would you be willing to share your (RP2040)-friendly door mod?

Sure, but fair warning, it is a hack, at best. My CAD skills are pretty basic and only having the STL to work with makes it harder, though I don't doubt someone skilled in CAD could do a better job than I did.

[modded-vented-sb-cover.zip](https://github.com/JaredC01/Galileo2/files/13795790/modded-vented-sb-cover.zip)

Ixian, 2024-01-01:
Just to update:

I was working on an improved vented cover when someone beat me to it - it's better than mine, though still not quite a perfect fit. You can find it here: https://www.printables.com/model/701255-galileo-2-cable-cover . Hopefully a better solution comes along when the Galileo 2 design is complete and the CAD files are released. 

On a related note, if you are using one of these sb boards with a Galileo 2 extruder I made a new cable guide/strain relief connector for it here: https://www.printables.com/model/701496-galileo-2-ebb-sb22092240-cable-and-connector-strai

JaredC01, 2024-01-02:
I've got this on my list of reworks prior to full release.  I'm likely going to have to flare the door out a bit to get the amount of clearance needed, but it shouldn't cause any issues visually or otherwise.

JaredC01, 2024-03-25:
Issue should be fixed with release.  Closing issue; please reopen if you find this still to be an issue.

closed 2024-03-25 by JaredC01