JaredC01/Galileo2 #27: EBB SB2209 CAN Prevents Fan cover to be flush

Shan723, 2023-12-29:
Hi I purchased a  EBB SB2209 CAN on Amazon but when installing the front cover fan it doesn't land lay flush to the front body and I think its due to board EBB SB2209 CAN sticking out more.
[image]
[image]
[image]

clau0086, 2023-12-31:
I'm having the same problem with the EBB SB2209 RP2040.  It's not as bad as yours.  I think moving the mounts back about 0.65mm would solve the problem.
[image]

Hulign, 2023-12-31:
I don’t have any problem with the SB2240 this is the twinsor body but I have tried a couple others with no issue. 
You could always heat up the inserts and move them over a smidge.

[image]

clau0086, 2024-01-01:
Hulign - I though your idea was worth a try and it worked.  The front fits flush and the USB port has been moved a reasonable distance from the stepper.  The holes opened up a bit too much and I had to fill in with epoxy to prevent the brass inserts from spinning.  It looks like I'm going to need to edit the STL's or wait for an official update.  Thanks!
[image]

clau0086, 2024-01-01:
[image]

clau0086, 2024-01-02:
[image]

clau0086, 2024-01-02:
[image]

JaredC01, 2024-01-02:
The board mounts are in the same location as CW2, so they should work as-is if they were working on CW2.  I will say, there's a bit of adjustment / flexibility in the board mount itself, so my suggestion is to leave the board mount screws slightly loose, push the SB front on, then tighten down the board with the front in place.  That should give the best chance of a good seat.

clau0086, 2024-01-02:
I wonder if BTT didn't locate the holes correctly.  It just didn't seem right as it was.

JKtech3D, 2024-01-03:
Everything lined up perfectly for me with almost no space between the USB and motor.

Shan723, 2024-01-04:
I redid this again and it all fits, leaving the screws loose helped

closed 2024-04-02 by JaredC01