JaredC01/Galileo2 #31: G2E ercf dual sensor

Wolffy00, 2024-01-05:
possibility to add dual sensor ?

evzenmayer, 2024-01-06:
I've added just one — the Toolhead Sensor (located behind the extruder wheel) by myself. The Entry sensor is not as necessary from my point of view, but possibly there is some place for it as well. It was rather tricky to find a suitable place, and as the CAD files are not published, I'm not too proud of the result. Editing reverse-engineered STL files in Fusion is not a pleasurable experience. That's why I'm not pushing this MOD to GitHub; instead, I am attaching it here if you want to give it a try or find inspiration for the author.

The embedded support needs some tuning, but I was able to remove it. I also haven't played any games regarding tolerances—just printed it out on my Ultimaker S5, and now I have a flawlessly working prototype in my Trident. You only need a 5.5mm steel ball, an Omron D2F microswitch, and two screws.

[Front_body_ECAS_Toolhead_Sensor.zip](https://github.com/JaredC01/Galileo2/files/13851546/Front_body_ECAS_Toolhead_Sensor.zip)

Wolffy00, 2024-01-06:
https://github.com/IRTrail/Voron-Stuff/tree/main/G2E%20Filametrix%20Mod/Twinsor

closed 2024-01-06 by Wolffy00