JaredC01/Galileo2 #34: Hall effect filament sensor

mcbeenb, 2024-01-14:
Sorry if this is being tracked else where, but is there a plan to include a hall effect sensor in the body?

ChaosBlades, 2024-05-07:
The community has moved away from hall effect filament sensors because they have been proven unreliable and started using micro switches with a ball bearing so trying to get someone to make a hall effect version probably isn't going to happen.

Already several Beta mods with filament sensors. Here is one with both pre and post gears.
https://www.printables.com/model/827300-galileo-g2e-ercf-2-sensors-with-step-datei

JaredC01, 2024-06-26:
Beta parts should NOT be used at this point, as there have been changes to the front body that make those parts incompatible with release G2E.  There are versions based on the release G2E that include filament sensors available from the ERCF group, please use those if needed.

closed 2024-06-26 by JaredC01