JaredC01/Galileo2 #36: [Feature Request] G2E with filament sensor options

Surion79, 2024-02-01:
Kind request for a G2E version with 2 sensor capabilities, so G2E can be used with ERCF.
I found an unofficial reverse engineered version, but i prefer an official one.

Thanks :)

karlfife, 2024-02-01:
I'm wondering if a better official solution would be a filament encoder in the G2E versus a filament sensor.   Encoders and Sensors are similar in size/weight/complexity, but an encoder detects more filament issues (i.e. jams/misfeeds, not just runout).  Klipper integrates encoders seamlessly using the same GPIO pin.  Could be that it's  not feasible to incorporate an encoder into the specific available space of the G2E, but in my opinion it would be a good evolution over a binary sensor.  Furthermore, the extruder is the correct placement for an encoder since it eliminates the possibility of a false positives due to failure to 'take up the slack' between the encoder and the extruder before starting/resuming a print.

Surion79, 2024-02-01:
You could place that request as an independant Feature Request. But it doesn't work with the mechanics of the ERCF, who has already an encoder and it requires for best results to know when the filament is entering the toolhead and when it is past the gears.
I request a slight change to add two omron switches. you suggest to rework this project completely.

karlfife, 2024-02-01:
A separate feature request is a good suggestion.  
As far as filament changers go, an encoder at the toolhead also functions a filament sensor.  The filament arrives when an encoder state change occurs.

Surion79, 2024-02-01:
if i see the available encoders i am not convinced that two would fit in the G2E to fulfill the requirement for the ERCF. Since the ercf already has an encoder, i don't see the use for a 2nd or even third.

qqqwqk, 2024-02-01:
> I'm wondering if a better official solution would be a filament encoder in the G2E versus a filament sensor. Encoders and Sensors are similar in size/weight/complexity, but an encoder detects more filament issues (i.e. jams/misfeeds, not just runout). Klipper integrates encoders seamlessly using the same GPIO pin(s). It could be that it's not feasible to incorporate it into the specific available space of the G2E, but in my opinion an encoder would be the correct evolution over a binary sensor. Furthermore, the extruder is the correct placement for an encoder since it eliminates the possibility of a false positives caused by slack between the encoder and the extruder when starting/resuming prints.

Due to the presence of the motor, the gear on the toolhead cannot act as an idler gear to read the filament status.  What we need is a bool sensor (eg: Omron d2f) to determine the filament has arrived or not to control the extrusion motor on the toolhead

evolwun, 2024-02-02:
Agree! I would love to see a filament sensor option. If one is not planned, I will start working on one as soon as CAD is released.

JaredC01, 2024-04-02:
ERCF has solutions for these already.  Closing.

closed 2024-04-02 by JaredC01