JaredC01/Galileo2 #37: Part broke during filament loading/tightening

sharpner, 2024-02-06:
I had this part break for me either during filament loading or maybe I tighten the screw/wrong too much?
[image]

It then clogged the filament path until I cleaned it up.
[image]

But I just screwed the g2e together and I am printing fine - so I wonder if that could be improved without actually holding the bearing with a small wall.

sharpner, 2024-07-29:
[image]

And it happened again... am I the only one? Am I doing something wrong?

I have no clue what's going on here - will test to print that part now in ASA

sharpner, 2024-07-29:
I wonder how this should be sliced, with extrusion width of 0.45 its only two walls
[image]
but with extrusion width of 0.4 its three
[image]

JaredC01, 2024-07-29:
This has happened on a few occasions, though I'm not 100% sure of the cause yet.  You can adjust the extrusion width and test, and please let me know the results.  I can't make a "fix" to the issue until I understand the cause, so any help with that would be appreciated.

sharpner, 2024-07-31:
I understand, I am just happy that I am not the only one.

Will let you know if it breaks again, last time it took 4-5 months :D 

I printed the part in ASA with 0.5 extrusion width so it resulted in exactly 1mm wall