JaredC01/Galileo2 #38: The g2_ptfe_cutting_tool.stl does not seem correct

sundansx, 2024-02-08:
I am assembling a stealthburner/tap/Dragon toolhead and I cut the PTFE feed tube the length specified by the G2 jig.  I then tested the filament path and kept getting hangups.  I found that I could push that PTFE tube farther up into the G2 channel and they went away.  After comparing the jig created PTFE tube and jig to the drawings, I am not sure how this could work.  I did my own measurements and ended up making a tube that was 46mm long and it fit right.  No more hangups.  I am using G2 with the Phaetus Dragon hotend parts.  See included drawing.

[image]

sundansx, 2024-02-08:
user error.

closed 2024-02-08 by sundansx

JaredC01, 2024-02-08:
No worries!  I appreciate the thoroughness of your initial post at least, that made finding the issue a lot easier!