JaredC01/Galileo2 #41: Cad files

izanamisu, 2024-02-29:
We are waiting for the cad / step files for us to start modding for ercf etc. 

People starting to reverse engineering it. When are the cad files coming out?

JaredC01, 2024-02-29:
As soon as I can get them out, they will be out.  I've been short on time for the past few months, but I am still actively working on this.  I've had a bit more time in the last week or so, and so long as I'm able to keep pace, they should be out in the next couple of weeks.

ChaosBlades, 2024-03-03:
@grimlokason I know this is off topic but those POGO pins are only rated for 2A which at 24v is only 48 watts. So you are going to be stuck at 40w heater which isn't enough anymore unless you want to print at original Ender 3 speeds. Also they add  "Contact resistance: 20mW max" which I think might throw off thermistor readings. For the front cover of the Stealthburner will be fine but the hotend, no way.

>  Current rating: 2A (continous), 3A (peak) per contact

https://www.mill-max.com/catalog/download/2019-08%3A014.pdf

grimlokason, 2024-03-03:
Hello @ChaosBlades 

for the power concern : it's in the [Mounting options](https://github.com/falcon14141/Voron_Mods/tree/master/Stealthburner%20POGO%20Toolhead#mounting-options)  we can use 2 pogo for the hotend. but I'll try to do a PR to the doc with your feedback to give more information.

About your concern for the thermistor, i'll check with the one who have done the mod.

Thank you !

( and sorry izanamisu, JaredC01 for the Hijack >.< )

falcon14141, 2024-03-03:
> @grimlokason I know this is off topic but those POGO pins are only rated for 2A which at 24v is only 48 watts. So you are going to be stuck at 40w heater which isn't enough anymore unless you want [...]

Current rating is indeed a real concern.. 
As stated by grim lokason the pogo mod was design was an option to double the pins used for extruder power... providing a 80w option with 100%pwm (120w peek power).
Return of experience is very positive, if using the millmax pogos.... Hartk tried some cheap ali pogos which didn't last very long.
My setup has been woking since the release date. Do the pogos induce a drift to the temp reading? This is indeed a nice question, I haven't check or measured, but if been printing my ASA with a 250° setting and have great quality. Anyway...  filament settings always need to be tuned ....

JaredC01, 2024-04-02:
CAD released, and ERCF team already released updated STLs.  Closing.

closed 2024-04-02 by JaredC01