JaredC01/Galileo2 #45: JosAr/G2_Umbilical_for_z_chain is not compatible with SB2209/SB2240

luxflow, 2024-04-02:
hi @jlas1 

SB2209/SB240 can male connector interfere with PG7 
this is because bigtech can female connector is top mount while SB Toolhead PCB, FLY-SB-2040 one is back mount

[image]

jimgineer, 2024-07-11:
I have this problem too. I thought 2209/2240 canbus boards were a shoe in for a mount. I don’t have time to design one myself and can’t seem to find one that works. I love the extruder it’s great but mount options need help.

jlas1, 2024-07-15:
Can you try this version
[Galileo_2_SB2209_2240.zip](https://github.com/user-attachments/files/16238124/Galileo_2_SB2209_2240.zip)
[image]
it uses a ziptie to secure the cable, as the gland is no longer necessary.

jimgineer, 2024-07-16:
I just bent the cord like that first pic. Thanks for posting those btw! I didn't picture it working like that. In my mind i was picturing something along the lines of the cw2 mount. Where the cord is protected from coming knocked loose or from an unforeseen snag.