JaredC01/Galileo2 #48: Cablechain hitting extruder motor

Exeu, 2024-04-04:
Hi there!

I recently upgraded to an Galileo 2 extruder on a ldo voron 2.4. 

Since then I lost 5-7 mm on x since the cable chains are hitting the extruder motor preventing it from moving further.

My guess is that the cable chains are too big?
Is there any recommended cablechain setup?

Thanks for an answer 
☺️
[image]

stramius, 2024-04-16:
Hi,

Unfortunately, I am facing the same issue. 

Is there any Workaround or mod for this?

Exeu, 2024-04-17:
@stramius what i did now is to redo the cable chain on x, with the hope that on a slightly different position it will work.
It got better now but it is still hitting the motor, but overall i were able to reduce it to 1-3mm max axis loss.
To solve this i guess the only way is to set the motor 1-2 higher or try to get "slimmer" XY Joints (they are causing the chain to move up at that position)

stramius, 2024-04-17:
@Exeu : Thank you so much for your Input. I have changed the cable chain position and now it is slightly improved. Anyway, it seems as a long term solution I need to switch to a Canbus in oder to get rid of the cable chains.