JaredC01/Galileo2 #50: G2SA missing PTFE information

Cova, 2024-04-05:
There is no information anywhere in the G2SA assembly manual that a piece of PTFE should even be installed, let alone how much should be sticking up out of the toolhead/hot-end to insert into the G2SA bottom.  It wouild be great to not just have the required PTFE length documented (I believe it is 6mm - based on using the Measure tool in slicer), but also have a PTFE cutting tool model that you can just set on top of the hot-end and cut on a line.

damianosullivan, 2024-04-26:
Can anyone confirm the ptfe length required?

ChaosBlades, 2024-05-07:
If you have something to measure the PTFE tube to cut it to the correct length then you have something to measure the hole depth.
With that said the STL in CAD measures 6mm
[image]

JaredC01, 2024-06-26:
Different applications will require different lengths, and as there's no "standard" configuration, there's no standard PTFE length.  ChaosBlades is correct though, the hole is 6mm deep in the G2SA body.

closed 2024-06-26 by JaredC01

thetic, 2024-06-29:
> the hole is 6mm deep in the G2SA body.

I think this is worth noting in the manual.