JaredC01/Galileo2 #52: scratchy noises from Galileo 2 Z

tofinnsch, 2024-04-20:
I have a slightly scratchy squeaking noise when the Z axis is shutting down, what could that be?
I took note of the shims in the planetary gear, which are missing from the instructions

tofinnsch, 2024-04-20:
I have now dismantled the Z-axes again and noticed that the countersunk screws cannot be screwed in completely flush. I will now heat these up and melt them flush. Hopefully the planetary gear will no longer rub against the screws, it already has slight scratch marks.
What is actually the opinion on lubrication in the gearbox?

tofinnsch, 2024-04-20:
Now I have removed all the axles again, melted the screws flush and greased the gears. it works without any unpleasant noises. Maybe the countersinks need to be made a little larger or the screws need to be screwed in with a melt thread

Hope it´s okay to post it here for somebody with the same issue

TamaBaka, 2024-08-24:
Ohh...I should've tried that.  I hated those countersunk screws because the rotor could and did catch on them.  When it did, the rotor was blocked and unable to continue turning.  Inserting a thick shim so that the rotor would peek over the screw heads didn't help because then it would tighten the assembly to the point that frictional forces would hard lock the rotor in place and prevent it from spinning.

I was expecting for that central piece to be thickened in a later CAD revision so that there would be room to recess the countersunk screws, but no changes have been made to the design since the beginning.  No advice on the size of the shim that's supposed to be there either.

QuinFromCanada, 2024-08-27:
I had the same issue, but didn't notice your post (issue #71 ) .  I ended up printing a small 0.4mm spacer as a temporary fix.  The post shows a possible cause of the issue. If you're interested.

Cheers.

TamaBaka, 2024-08-27:
> I had the same issue, but didn't notice your post (issue #71 ) . I ended up printing a small 0.4mm spacer as a temporary fix. The post shows a possible cause of the issue. If you're interested.
> 
> Cheers.

Looking at your issue...it never occurred to me that the layer height (0.2) that I was printing at could cause rounding errors. I can see that making things difficult.

QuinFromCanada, 2024-08-27:
> > I had the same issue, but didn't notice your post (issue #71 ) . I ended up printing a small 0.4mm spacer as a temporary fix. The post shows a possible cause of the issue. If you're interested.
> > Cheers.
> 
> Looking at your issue...it never occurred to me that the layer height (0.2) that I was printing at could cause rounding errors. I can see that making things difficult. Whatever the case, I had to loosen the screws too so it wouldn't bind with the shim in, but it annoyed me a lot that this meant that I couldn't tighten the housing screws securely.

My thought is that between the rounding error of the slicer and a bit of over-extrusion it easy to be out by at least 0.2.  Unfortunately tolerances for 3D prints is 50% skill, 25% luck, 25% black magic.