JaredC01/Galileo2 #53: g2sa tensioner arm screw

bosbmind, 2024-04-23:
Howdy! the g 2sa tensioner screw seems to not hold itself on the arm and will fall off the arm itself maybe some thin slots that cup around the sides of the spring capture on the screw so that way it doesnt fall off during filament changes

ChaosBlades, 2024-05-07:
You are saying you completely unscrew the tensioner when changing filament? It has a little bump so you can press on it with your thumb to compress the spring. I actually find this unnecessary though. Simply get the filament to the gear then turn the nob manually to feed it in. You should not have any reason to ever completely unscrew the tensioner unless you get a jam in the extruder. It should be tightened to the correct tension then left alone.

intplus10, 2025-03-16:
I have the original LDO kit and that spring is so strong I need to unscrew it fully. Filament can be changed with pressing the bump, but that putting unnecessary stress on the whole extruder. I can see everything bending (Archetype Blackbird). Also the heat insert from the housing came out due to the forces. Maybe the hole tolerance was a little bit big, but the wall on the top side is so thin it is bulging out when I press the (voron) heat insert. 

So if I need to unscrew it like 80% to reduce the forces it would be better to unscrew it fully without thinking on the thread length. Just turning the knob until its loose, and not turning it fearing that it will fall out any moment :)

Turning the knob is not everytime possible. If the macros for filament change are not including extruder off for example.