JaredC01/Galileo2 #54: Why is the Galileo2_extruder designed with a single-sided meshing gear

MethodJiao, 2024-04-30:
Why is the Galileo2_extruder designed with a single-sided meshing gear clamping the Filament, which often causes my extruder to slip. I also tried tightening the clamping screws, but this would cause the motor to stop rotating

ChaosBlades, 2024-05-07:
It is like the Nextruder from Prusa but more compact. Galileo 2 is not about grip it is about extrusion consistency. Dual gear extruders will never provide the extrusion consistency of single gear. Some of my printers I want extrusion consistency over grip (v0/Switchwire) so I use the Galileo 2 but other printers I need grip (v2.4 w/ ERCF / 10KG spools) so I go with HGX 2.0.

Nextruder - https://youtu.be/AaX1v6qWOnc?si=BXU3idvpRMim6Xeq
HGX 2.0     - https://youtu.be/l5SrNxmm5hk?si=dOtoFpsBQKG4Q_Iz

JaredC01, 2024-06-26:
As ChaosBlades implies, there's less overall grip with G2, though tension also makes a difference, and your assembly of the extruder makes a difference (especially if you don't align the gearbox properly).  Depending on the hotend, you should expect to get up to around 70 cubic mm/s of flow with a good quality pure ABS filament and a hotend that matches the capability.

If you need an extruder to push filament through what your hotend can handle, CW2 or any of the BMG-guts extruders will provider higher pushing forces (with other trade-offs).

closed 2024-06-26 by JaredC01