JaredC01/Galileo2 #56: G2zxl fitment issue

whezzel, 2024-06-01:
G2zxl kits are starting to become available at a few vendors. I ordered some kits for my 350 V2, and started to assemble them today.

I don't see an assembly manual in the repo for the g2zxl, so I used the manual for the standard g2z. On page 29 of the manual, it says to insert the 2x M5x10 screws used for mounting it to the frame. Unfortunately, one of the screws is not accessible once the stepper motor and gearbox are installed. 

Am I doing something wrong? Is there an assembly manual for the xl kit? Does the xl kit have to be attached to the frame before installing the stepper and gearbox?

[image]

JaredC01, 2024-06-01:
You ARE doing something wrong, but in this case it's just order of operations.  G2ZXL requires a different assembly order, and unfortunately the manual is not ready yet.  I DID just put a video together today though, so good timing.

https://youtu.be/_yT3NV2jkFM

whezzel, 2024-06-01:
Awesome. I figured it was an order of operations thing. I was wondering if it needed to be mounted to the frame before installing the stepper and gearbox. Now i know. Thanks for the quick response. Ill go ahead and close this issue.

closed 2024-06-01 by whezzel