JaredC01/Galileo2 #58: Feature Request: G2S Filament Sensor

KlausHans, 2024-06-16:
Hello,

a filament sensor for the G2S would be great. There exists a model for the G2E on printables, but nothing for G2S. 
Something similar to the orbiter 2 sensor would be great, i think.
Thanks.

Bb06477, 2024-06-17:
https://www.printables.com/model/915344-g2sa-orbiter-2-mount-main-frame-with-a-filament-se

try this UNTESTED

Bb06477, 2024-06-18:
https://www.printables.com/model/915344-g2sa-orbiter-2-mount-main-frame-with-a-filament-se

Try this. I just made it and I'm no expert (nor am i associated with the Galileo 2 Project), its mostly untested. but Ive printed it a couple times and seems pretty reliable in terms of hitting the switch. This is going into my G2SA to XOL rapido 2 uhf, w ERCF build so i may add a tool head sensor as well if needed.

JaredC01, 2024-06-26:
G2SA will be undergoing a design refresh to bring it more in line with G2E and G2Z on the gearbox side of assembly.  I likely will NOT be incorporating a filament sensor into the design, but once I have the design finalized, some of the other groups can work on adding the sensor.

If you need something now, Wristwatch G2 has the option for dual filament sensors and uses the G2E guts just like G2SA does, it's just in a different package.

https://github.com/bythorsthunder/Voron_Mods/tree/main/Wristwatch_G2_Dual_Filament_Sensor
https://github.com/tetsu97/WristWatch-G2-Extruder

closed 2024-06-26 by JaredC01

KlausHans, 2024-06-26:
Thanks for pointing out the WristWatch G2 extruder, i didn't know this exists.

diabolicaldan, 2024-12-02:
I just saw that WristWatch files were updated, thought I would share here too:
https://github.com/tetsu97/WristWatch-G2-Extruder/tree/main