JaredC01/Galileo2 #70: G2Zxl backlash issue

roadkill667242, 2024-08-17:
i have tried 3 different sets of new gears and all have a backlash issue in a specific spot. I have made sure the motor slides in easy and isnt binding. reprinted the gear carrier. done everything but reprint the main housing but that shouldnt matter for how the backlash is. please advise.

JaredC01, 2024-08-17:
I need more info... have you used G2Z prior to G2ZXL, and if so, did you have issues with G2Z as well? The gears are all the same, the motor is the only thing different between G2Z and G2ZXL kits, and the motor has the same spur gear spec as G2's NEMA14 stepper.

If you can get a video of the issue, that would be ideal.

roadkill667242, 2024-08-18:
New install from belt drive.

roadkill667242, 2024-08-18:
https://github.com/user-attachments/assets/e9bb1e35-1a06-4232-a3ef-cfe7ca3582b8

roadkill667242, 2024-08-18:
The oddest thing about it. It's only in specific spots. It will go to no archmin to maybe if i have to guess 30 archmin.

roadkill667242, 2024-08-18:
I think I sourced it.  I will report back after further test

woofermazing, 2024-09-09:
Did you figure this out?