JaredC01/Galileo2 #73: Filament gets stuck during unload

husnainali09, 2024-08-30:
Have printed two G2E SB to be sure but during unload filament gets partially stuck right at the very end and needs a hard tug to come out completely.

I believe this is just after the gear. My filament on unload is a bit fatter at the tip.

Can't figure out why (I did the 2mm drill bit)

Maybe my unload macro is not optimal.

Can you share your unload macro for me to test please.

husnainali09, 2024-10-01:
Is this project abandoned ? 

Requesting unload lengths for ideal normal and UHF please

JaredC01, 2024-10-21:
Hi @husnainali09!  This project is not abandoned, but it is currently stale due to my real-life schedule.  If you're having issues with unloading, your issue will be with your hotend and/or your unload macro.  If your filament is "fatter at the tip" then it sounds like you either need to retract a bit faster initially, or swap nozzles and test again.

pterodaktil02, 2025-05-04:
I encountered the same issue.

I don’t know how you resolved it, but the following version of the unload macro helped me:
`[gcode_macro UNLOAD_FILAMENT]
variable_ochistka_distance: 25  ; расстояние очистки 25 мм
gcode:
    {% set speed = params.SPEED|default(300) %}
    {% set max_velocity = printer.configfile.settings['extruder'].max_extrude_only_velocity * 60 %}
    {% set max_temp = printer.configfile.settings['extruder'].max_temp|float %}  ; максимальная температура экструдера
    {% set original_temp = printer.extruder.target|float %}  ; сохранить текущую температуру
    {% set default_temp = 250 if original_temp == 0 else (original_temp + 10) %}  ; 250°C или текущая +10°C
    {% set filament_temp = params.TEMP|default(default_temp)|float %}  ; использовать заданную или вычисленную температуру
    {% set safe_temp = [filament_temp, max_temp]|min %}  ; ограничить температуру до max_temp
    M109 S{safe_temp}  ; нагрев до безопасной температуры
    SAVE_GCODE_STATE NAME=unload_state
    G91  ; относительный режим
    G92 E0  ; сброс экструзии
    G1 E{ochistka_distance} F{speed}  ; очистка 25 мм
    G4 S3  ; пауза 3 секунды для стабилизации
    G1 E-60 F900  ; начальные 60 мм на скорости 15 мм/с
    G1 E-120 F{max_velocity}  ; оставшиеся 120 мм на максимальной скорости
    G4 S5  ; пауза 5 секунд для дополнительного охлаждения
    RESTORE_GCODE_STATE NAME=unload_state`