JaredC01/Galileo2 #74: G2ZXL Binding

Kbreezy24, 2024-09-05:
Hi @JaredC01 

Thanks for your awesome contribution with the G2!

A customer of ours was having some binding issues with his assemblies so we had them sent to us for evaluation. Here's what we found:

The print tolerances versus cad were confirmed good, but when the gear train stack was fully tightened using nuts and bolts (versus the motor being present) the planetary system would bind. This is obviously true when the motor is present as well, but the super high gear reduction makes it tough to feel for.

In our case increasing the flange height of right_stepper_plate_G2zxl by 0.4mm allowed the POM ring gear to be flush with the flange height IRL - which solved the binding issue. 

[image]

This may or may not be related to issue #71 and would also likely affect G2Z and perhaps G2E

erlis, 2024-09-22:
Just confirming, I was having a click click click on my XL and I wasn't sure if it was binding, I did this mod on the printed part and it's gone, so I guess I was having some binding after tightening the screws, it's very hard to tell because you can rotate the shaft to "feel" it. 

So +1 to this

mooseroy, 2025-03-15:
I was having a lot of binding with the stock plate, so printed the modified version from Kbreezy24, and now they are super smooth. 
+1 to this mod.

Steiner-SE, 2025-05-18:
> I was having a lot of binding with the stock plate, so printed the modified version from Kbreezy24, and now they are super smooth. +1 to this mod.

Do you have  a link for this fix please?

Kbreezy24, 2025-05-20:
[G2ZXL_ModifiedFlange.zip](https://github.com/user-attachments/files/20320441/G2ZXL_ModifiedFlange.zip)

Here you are!

> > I was having a lot of binding with the stock plate, so printed the modified version from Kbreezy24, and now they are super smooth. +1 to this mod.
> 
> Do you have a link for this fix please?

Steiner-SE, 2025-05-20:
> [G2ZXL_ModifiedFlange.zip](https://github.com/user-attachments/files/20320441/G2ZXL_ModifiedFlange.zip)
>
> Here you are!

Thank you kindly!
Out of curiosity, why only the right one, why not all four?

Kbreezy24, 2025-05-20:
> Thank you kindly! Out of curiosity, why only the right one, why not all four?

This was just a confirmation print that the issue was solved. If the parts aren't the same for all 4 corners then you should be able to mirror this STL. Happy to help further if not, though!

Steiner-SE, 2025-05-20:
> > > [G2ZXL_ModifiedFlange.zip](https://github.com/user-attachments/files/20320441/G2ZXL_ModifiedFlange.zip)
> > > Here you are!
> > Thank you kindly! Out of curiosity, why only the right one, why not all four?
> 
> This was just a confirmation print that the issue was solved. If the parts aren't the same for all 4 corners then you should be able to mirror this STL. Happy to help further if not, though!

Think they are mirrored, and they are marked with those numbers "1" and "2" to line up with the other part.

Steiner-SE, 2025-05-20:
Oups, never mind, confused myself lol. Thank you again.

Frigolol, 2025-05-20:
https://www.printables.com/model/1301950-g2zxl-binding-fix