JaredC01/Galileo2 #75: Gear Ratio Question: Isn't It 8:1 instead of 9:1?

ecoinomist, 2024-09-06:
So I counted the gear teeth as illustrated in the manual. There 9 teeth gear for the motor shaft, which drive the three inner gears with 31 teeth, that in turn spins the housing with 72 teeth. This results in:

72/31/9 = `8:1` gear ratio

Have I miscalculated something, or has it been overlooked? The docs says to put `9:1` gear ratio into printer config.

JaredC01, 2024-09-06:
Your calculation method is incorrect.  While the number of teeth on the sun gear and number of teeth on the ring gear are correct, you're forgetting the rotation of the carrier itself.

The easiest way to calculate the ratio for this type of gearbox is: (Ring Gear Teeth / Sun Gear Teeth) + 1.  That results in a 9:1 ratio for Galileo 2.

ecoinomist, 2024-09-06:
Thank you, I was just watching this on [Youtube](https://youtu.be/71dn-eVdSmc?si=_YFhaxXaaIvU7yAP&t=1022), and indeed that's the case.

closed 2024-09-06 by ecoinomist