JaredC01/Galileo2 #76: G2SA release? Any help needed?

ChaosBlades, 2024-09-06:
It has been 7 months since any changes were made to G2SA. I have had no issues that entire time. Been waiting for CAD so I could make cosmetic changes. MMU mostly.

@JaredC01 do you need any tests ran or help with CAD finalization to get this through the finish line?

JaredC01, 2024-09-07:
It's been longer than expected, yes.  I broke my pinky on my dominant hand back in March and have been playing catch up with life since then.  G2SA is still getting reworked enough that the current version will be obsolete, so I'm not likely going to push the current CAD, but I understand the want for updates / upgrades to the existing platform.

Send me a DM on Discord and we can talk about it.

ChaosBlades, 2024-09-10:
I sent a message on Discord from an account called Akanar.

filamonkey, 2024-10-11:
I had posted a new issue with missing CAD, but as I understand it won't be coming out for this version. I am planning on replacing two Titan extruders with the G2SA. Will the form factor change a lot?

JaredC01, 2024-10-11:
> I had posted a new issue with missing CAD, but as I understand it won't be coming out for this version. I am planning on replacing two Titan extruders with the G2SA. Will the form factor change a lot?

It's not likely to change much in the overall form factor, no.  At this time, I will be "passing the torch" to @hartk1213 to push this to the finish line, as I'm still absolutely swimming in work needing to be finished.

ecoinomist, 2024-10-25:
Also please ensure the kit release has stepper motor cable length of only 5mm. Currently, the cable length of the G2E is too short to wire to the electronics compartment, and too long to tuck away in the toolhead.

swatkxs, 2025-03-13:
The current G2SA worked fine for me. And waiting for CAD.