JaredC01/Galileo2 #80: Idle Clogs

Biffes, 2024-10-27:
Hello,
I have an unusual problem with the G2E and random clogs. I'm currently using a very "soft" PLA filament (Noname). It happend with "matte" filament too. My spring tension is at an absolute minimum. 

The extruder has now caused blockages several times at the start of printing. After a few tests, I found that the blockages occur when a print was completed shortly before, i.e. the print head still had residual heat. The residual heat in the printer is not particularly high (approx. 35 degrees). However, the extruder motor may radiate additional residual heat. 

I have now always pulled out the filament after about 15 minutes of finished prints and you can see that the pressure of the ball bearing unfortunately crushes the filament across its width and thus causes blockages.
[image]
[image]

 
In old Bowden extruders, this was dealt with by installing U-ball bearings (or groove bearings) in the extruder, thereby at least maintaining the shape of the filament.
[image]

Unfortunately I couldn't find a ball bearing in the right size.

Is this issue known? Are there any fixes?

If necessary, you could simply add ventilation channels to the extruder gear in the printer to drop the tempreature. Slightly moving the ball bearing could also help.

ChaosBlades, 2024-10-27:
Reducing stepper run current would help.

So would one of those stepper fan cooler mods.
https://www.orbiterprojects.com/turbiter/

Also retracting filament by 10mm at end of print may help with extruder heat creeping up the filament.

Biffes, 2024-11-09:
Hello, thank you for your help.
I followed your suggestions and found a different solution..
The problem with the extruder is that the drive shaft is probably made of aluminum and transports the heat from the motor.
[image]

I did some tests with a flir camera and had a 50 deg. temp in the extruder. 

In addition, the temperature of my extruder motor rises (!) when it is in the waiting state after the finished print (+~5-10degrees). This is probably because Klipper no longer recommends "hold_current" and the case fans stop operating when the hotend temperature falls below 50 degrees. 

I therefore set a hold_current of 0.3 again and the temperature of the motor drops significantly again after printing is completed. 

I also tested the "Turbiter". This is very effective, but unfortunately it causes vibrations on the print head. 

Thanks for the help!

agravelot, 2024-11-09:
> Hello, thank you for your help.
>
> I followed your suggestions and found a different solution..
>
> The problem with the extruder is that the drive shaft is probably made of aluminum and transports the [...]

You can disable your extruder stepper at the end of the print