JaredC01/Galileo2 #89: Jamming when loading filament.

LazaroFilm, 2025-04-26:
The G2E works really nicely when printing but loading the filament is always an issue with possible jam, forcing a full disassembly of the extruder. 
I feel like the channel after the gears isn't properly aligned and with the angled entry, the filament gets pushed to the side.
I wonder if adding a small chamfer to the table entrance after the gears would help. 

[image]

LazaroFilm, 2025-05-07:
Not yet. I printed a new g2e with filamatrix but my thermistor broke and I'm waiting for a new one.

Karl-Barx, 2025-06-22:
> I'm having the same issue since one month. It'll depend of the shape of the filaments. If curved, it'll failed if not correctly oriented. If straight, it seems to be ok. Nb : will disasembling for checking, i noticed that my drive gear was loose. I hope it'll solve the issue to screw it again with threadlocker !

Did that solve your issue? Ive tried retightening everything, and printing the parts again and im still having the same issue. You're right about the shape of the filament it makes a huge difference if there is any curvature while trying to load everything

LazaroFilm, 2025-10-17:
On my end I used a ceramic exacto and cleaned the entrance of the filament after the gears and it has solved this problem. I'm now off to fix to all the other problems I'm facing.